ipxe/ipxe #1472: HP Proliant DL360 doesn't obtain IP address and does not run embedded script

kbmetz, 2025-05-22:
Trying to use iPXE and UEFI boot on a HP Proliant DL360 Gen9

I've compiled ipxe as follows and copied it to the server with the name 'custom-ipxe.efi'

make bin-x86_64-efi/ipxe.efi DEBUG=device:1,efi_driver:1 EMBED=demo.ipxe

demo.ipxe:
#!ipxe
prompt --key 0x02 --timeout 4000 Press Ctrl-B for the iPXE command line... && shell ||
echo Running custom script
ifstat
 dhcp
chain http://10.14.165.11/ipxe/boot.ipxe

The debug doesn't offer any clues and if I turn the verbosity up it scrolls off the screen and is unreadable. But more importantly I never get the option for a console. It does not appear that the demo.ipxe commands are run

I'm using dnsmasq and have confirmed that it is calling custom-ipxe.efi as noted in the screenshot

I've included demo.ipxe in my tftp root and I've also made a autoexec.ipxe file. The autoexec.ipxe has the same commands as demo.ipxe

Server boots bootx64.efi and will bring up a grub menu with no issue.

[image]

NiKiZe, 2025-05-22:
Is this with current git master?

Have you tried building snp.efi or snponly.efi instead?
There is several other issues and discussions already for very similar or identical experiences.

kbmetz, 2025-05-23:
I pulled from source about two weeks ago. I'll update again.

I've tried building snp.efi, it did the same thing. Booted and timed out waiting for autoeec.ipxe. Tried booting snponly.efi and it didn't even show any text, just cycled through the interfaces and didn't appear to load at all.

joernheissler, 2025-09-17:
iPXE on DL360 Gen9 in UEFI mode (secure boot turned off) works for me.

> if I turn the verbosity up it scrolls off the screen and is unreadable

Try adding a serial console. It's available through SSH on the iLO.