ipxe/ipxe #1485: 10ec8136 linking error

tails1154, 2025-06-28:
[image]

[image]

[image]

refer to the images.

Thanks for any help! :D

tails1154, 2025-06-28:
lcpci -nn
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

tails1154, 2025-06-28:
sorry

NiKiZe, 2025-06-28:
When building, you should be able to use copy paste.
Most of the images are not readable, often cut of.

What I can find is 
`Could not find isolinux.bin`
If you do need the iso, make sure you have installed the required dependencies for that, if you don't need iso, use a different build target. Have a look at #1118