ipxe/ipxe #1515: build with gcc15 `-ffat-lto-objects` fails for `arch/x86/interface/pcbios/int13.c`

TheJJ, 2025-08-13:
hey! this may well be a compiler bug, but maybe it's still something to address in ipxe :)

version 5b3ebf8b24ae40a6f9f9f78491702d508f843e56

gcc version `15.1.0` or `15.2.0-1ubuntu1`

build fails with ` -flto=auto -ffat-lto-objects`
```
x86_64-pc-linux-gnu-gcc -DARCH=i386 -DPLATFORM=pcbios -DPLATFORM_pcbios -DSECUREBOOT=0 -march=i386 -fomit-frame-pointer -fstrength-reduce -falign-jumps=1 -falign-loops=1 -falign-functions=1 -mpreferred-stack-boundary=2 -mregparm=3 -mrtd -freg-struct-return -m32 -fshort-wchar -Ui386 -DNVALGRIND -fno-stack-protector -Iarch/i386/include -Iarch/x86/include -Iinclude -I. -Os -g -ffreestanding -fcommon -Wall -W -Wformat-nonliteral -Wno-array-bounds -Wno-dangling-pointer -fno-dwarf2-cfi-asm -fno-exceptions  -fno-unwind-tables -fno-asynchronous-unwind-tables -Wno-address -Wno-stringop-truncation -Wno-address-of-packed-member -Wno-unterminated-string-initialization -fno-PIE -no-pie  -fcf-protection=none  -ffunction-sections  -include include/compiler.h -DASM_TCHAR='@' -DASM_TCHAR_OPS='@' -Ulinux   -DOBJECT=int13 -c arch/x86/interface/pcbios/int13.c -o bin-i386-pcbios/int13.o -flto=auto -ffat-lto-objects
In file included from arch/x86/include/realmode.h:7,
                 from arch/x86/interface/pcbios/int13.c:40:
arch/x86/include/librm.h:103:9: error: ‘*num_drives’ causes a section type conflict with ‘*rm_ds’ in section ‘.text16.data’
  103 |         _text16_ ## variable __asm__ ( #variable )
      |         ^~~~~~~~
arch/x86/include/librm.h:103:9: note: ‘*rm_ds’ was declared here
  103 |         _text16_ ## variable __asm__ ( #variable )
      |         ^~~~~~~~
```
when the lto flags are removed, the file compiles.

It seems weird that num_drives (as a `uint8_t`) is added to the `text16` segment. Is this correct?

mcb30, 2025-08-13:
Yes, it's correct that `rm_ds` is in the `.text16.data` segment.  iPXE needs some way to be able to locate the real-mode data segment when called via 16-bit entry points such as `pxe_entry`.  The PXE API does not define any way for the caller to pass in the callee's data segment: the callee (i.e. iPXE in this case) has to be able to determine its own data segment given nothing more than the current code segment and instruction pointer.  We therefore have to store `rm_ds` in the code segment, since until we read this value we have no idea where the data segment is.

For similar 16-bit INT 13 API-related reasons, `num_drives` is also stored so that it can be accessed while only the code segment is known.

iPXE doesn't support link-time optimisation, so I'm not entirely surprised that building with `-flto` will fail.  Build and link objects normally, and the problem will go away.

TheJJ, 2025-08-14:
Thanks for your insights :)
I've stumbled accross this while preparing the package update (likely `1.21.1+git-20250602.5b3ebf8b`) in Ubuntu for the upcoming questing quokka release: https://code.launchpad.net/~jj/ubuntu/+source/ipxe/+git/ipxe/+merge/490765

I've now disabled building with LTO.

Btw it would be great if you could do a new tagged release every now and then, that's easier to auto-check for updates for us 😄

NiKiZe, 2025-08-14:
> 
> Btw it would be great if you could do a new tagged release every now and then, that's easier to auto-check for updates for us 😄

See #544 and also https://github.com/ipxe/ipxe/discussions/679#discussioncomment-5214424
You might be able to help here by pushing for getting that shim signed thru any internal channels there might be. (To reduce the time spent waiting for MS to make progress)